dmazzella/ucrypto #1: [Question] Support for SECP128r1?

ekawahyu, 2021-08-03:
Will isupport for SECP128r1 is in the plan? Thank you.

dmazzella, 2021-08-03:
Hi,
you can change at the end of the file [modules/ufastecdsa/curve.py](https://github.com/dmazzella/ucrypto/blob/1349fa08481aa59c744123614fe5d0cd8804037b/modules/ufastecdsa/curve.py)

```python
P256 = Curve(
    0xFFFFFFFF00000001000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFFF,
    0xFFFFFFFF00000001000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFFC,
    0x5AC635D8AA3A93E7B3EBBD55769886BC651D06B0CC53B0F63BCE3C3E27D2604B,
    0xFFFFFFFF00000000FFFFFFFFFFFFFFFFBCE6FAADA7179E84F3B9CAC2FC632551,
    0x6B17D1F2E12C4247F8BCE6E563A440F277037D812DEB33A0F4A13945D898C296,
    0x4FE342E2FE1A7F9B8EE7EB4A7C0F9E162BCE33576B315ECECBB6406837BF51F5,
    name="P256",
    oid="2a8648ce3d030107",
)

SECP128r1 = Curve(
    0xfffffffdffffffffffffffffffffffff,
    0xfffffffdfffffffffffffffffffffffc,
    0xe87579c11079f43dd824993c2cee5ed3,
    0xfffffffe0000000075a30d1b9038a115,
    0x161ff7528b899b2d0c28607ca52c5b86,
    0xcf5ac8395bafeb13c02da292dded7a83,
    name="SECP128r1",
    oid="2b8104001c",
)

_CURVE_OIDS = {
    "2a8648ce3d030107": P256,
    "2b8104001c": SECP128r1
}
```

let me know if it's okay, i haven't tested the code i'm on vacation.

Greetings,
D.

dmazzella, 2021-08-03:
Hi @ekawahyu ,
thanks for your question, this has shown a bug (fixed in [b666c97](https://github.com/dmazzella/ucrypto/commit/b666c97f4bc8a321250c4ad493f44979ddac6e01)).

For your question a rapid solution can be edit [tests/fastecdsa_1.py](https://github.com/dmazzella/ucrypto/blob/master/tests/fastecdsa_1.py):

```python
try:
    from ufastecdsa import curve, ecdsa, keys, util

    get_bit_length = util.get_bit_length

except ImportError:
    from fastecdsa import curve, ecdsa, keys, util

    get_bit_length = int.bit_length

def main():

    SECP128r1 = curve.Curve(
        "SECP128r1",
        0xFFFFFFFDFFFFFFFFFFFFFFFFFFFFFFFF,
        0xFFFFFFFDFFFFFFFFFFFFFFFFFFFFFFFC,
        0xE87579C11079F43DD824993C2CEE5ED3,
        0xFFFFFFFE0000000075A30D1B9038A115,
        0x161FF7528B899B2D0C28607CA52C5B86,
        0xCF5AC8395BAFEB13C02DA292DDED7A83,
    )
    setattr(curve, "SECP128r1", SECP128r1)

    c = curve.SECP128r1  # curve.P256
    private_key = 129440090160411729250090482679899609177  # 82378264402520040413352233063555671940555718680152892238371187003380781159101
    public_key = keys.get_public_key(private_key, c)
    # private_key, public_key = keys.gen_keypair(c)

    print("PRIVATE KEY:", private_key)
    print("PUBLIC KEY:", public_key.x, public_key.y, public_key.curve.name)

    m = "a message to sign via ECDSA"

    r, s = ecdsa.sign(m, private_key, curve=c)

    print("R:", r, "S:", s)

    verified = ecdsa.verify((r, s), m, public_key, curve=c)
    print(verified)

if __name__ == "__main__":
    main()

```

the output with ufastecdsa:
```bash
➜  micropython (master) ✗ python -Bu tools/pyboard.py -d /dev/cu.usbmodem3267334E30372 ports/stm32/boards/PYBD_SF6/cmodules/ucrypto/tests/fastecdsa_1.py
PRIVATE KEY: 129440090160411729250090482679899609177
PUBLIC KEY: 28682380564611372386993306297071487642 241623269395742587426803653662887823675 SECP128r1
R: 205073144472716698218745305852045595857 S: 273838288313259806971983106551617826270
True
```
the output with fastecdsa:
```bash

➜  micropython (master) ✗ python ports/stm32/boards/PYBD_SF6/cmodules/ucrypto/tests/fastecdsa_1.py                                                      
PRIVATE KEY: 129440090160411729250090482679899609177
PUBLIC KEY: 28682380564611372386993306297071487642 241623269395742587426803653662887823675 SECP128r1
R: 205073144472716698218745305852045595857 S: 273838288313259806971983106551617826270
True
➜  micropython (master) ✗
```

ekawahyu, 2021-08-03:
I think I mistakenly look at your ucrypto with ucryptolib. I have never tried to build/compile micropython module, I will give it a try.

closed 2021-08-04 by dmazzella