dmazzella/ucrypto #2: Compiling for Micropython

TGiles1998, 2022-05-03:
Hi,

Do you have to change the make file to a cmake file in order for this to compile?

-- Adding linker script /Users/thomasgiles/esp-VS/micropython-3/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
Including User C Module(s) from /Users/thomasgiles/esp-VS/micropython-3/ports/esp32/boards/GENERIC_S3_SPIRAM/ucrypto
CMake Error at /Users/thomasgiles/esp-VS/micropython-3/py/usermod.cmake:42 (include):
  include requested file is a directory:

    /Users/thomasgiles/esp-VS/micropython-3/ports/esp32/boards/GENERIC_S3_SPIRAM/ucrypto
Call Stack (most recent call first):
  main/CMakeLists.txt:10 (include)

Many thanks,

Thomas

TGiles1998, 2022-05-03:
Command used:

make -C ports/esp32/ BOARD="GENERIC_S3_SPIRAM" USER_C_MODULES="/Users/thomasgiles/esp-VS/micropython-3/ports/esp32/boards/GENERIC_S3_SPIRAM/ucrypto"

TGiles1998, 2022-05-03:
Trying your example unfortunately doesn't work either 

make -j8 -C mpy-cross && make -j8 -C ports/stm32/ BOARD="PYBD_SF6" USER_C_MODULES="$(pwd)/ports/stm32/boards/PYBD_SF6/cmodules"
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /Users/thomasgiles/esp-VS/micropython-3/ports/stm32/boards/PYBD_SF6/cmodules/ucrypto
GEN build-PYBD_SF6/genhdr/qstr.i.last
Traceback (most recent call last):
  File "/Users/thomasgiles/esp-VS/micropython-3/ports/stm32/../../py/makeqstrdefs.py", line 196, in <module>
    preprocess()
  File "/Users/thomasgiles/esp-VS/micropython-3/ports/stm32/../../py/makeqstrdefs.py", line 70, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/multiprocessing/pool.py", line 870, in next
    raise value
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/Users/thomasgiles/esp-VS/micropython-3/ports/stm32/../../py/makeqstrdefs.py", line 54, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'arm-none-eabi-gcc'
make: *** [build-PYBD_SF6/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-PYBD_SF6/genhdr/qstr.i.last'

dmazzella, 2022-05-03:
I havn't a esp32 for test it and time to try it.

But any PR are welcome!

TGiles1998, 2022-05-03:
@dmazzella Thank you, can u provide a skeleton example for a cmake file? Im happy to do the rest and then open a pull request. I'm just not sure how to add a correct cmake file.
Many thanks!

dmazzella, 2022-05-03:
Look at [usqlite](https://github.com/spatialdude/usqlite)

TGiles1998, 2022-05-04:
Hi,

Does this seem reasonable?

Many thanks,

Thomas 

add_library(usermod_ufastecdsa INTERFACE)

target_sources(usermod_ufastecdsa INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}/moducrypto.c
)

target_include_directories(usermod_ufastecdsa INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}
)

target_link_libraries(usermod INTERFACE usermod_ufastecdsa)

dmazzella, 2022-05-04:
try with last push

dmazzella, 2022-05-04:
Please give me feedback on this, I have only tested the build.

MATTYGILO, 2022-05-04:
Going to start documenting errors from build:

```
if isinstance(h1, hashlib.md5):
    hash_size = 16
```

hashlib.md5 is not a thing

MATTYGILO, 2022-05-04:
Also in the case of hmac.py line 51 and line 65

MATTYGILO, 2022-05-04:
***ERROR*** A stack overflow in task mp_task has been detected.
When running fast_ecdsa1.py

MATTYGILO, 2022-05-04:
@dmazzella Are there any other ecc repos for MicroPython out there?

dmazzella, 2022-05-04:
> @dmazzella Are there any other ecc repos for MicroPython out there?

I don't know any others

MATTYGILO, 2022-05-04:
> > @dmazzella Are there any other ecc repos for MicroPython out there?
> 
> I don't know any others

Ok no worries I'm gonna try and work through it and get a working version.

dmazzella, 2022-05-04:
> Ok no worries I'm gonna try and work through it and get a working version.

thank you,
without hardware it is difficult for me to test porting to a new platform.

dmazzella, 2022-05-04:
> Hi @dmazzella
> 
> Unfortunately I could not get this to work, the python in this repo also needs altering.
> 
> Thanks
> 
> Thomas

native part work for you?

dmazzella, 2022-05-10:
I am closing this for now, in case of future updates, please feel free to reopen it.

closed 2022-05-10 by dmazzella