dmazzella/ucrypto #8: hashlib.md5() seems to be removed in latest uhashlib

y3k00000, 2022-12-13:
As title, found this issue when installing .py files to my pico_w.

I currently comment out the occurrences of hashlib.md5 to make it work.

y3k00000, 2022-12-13:
from [Document](https://docs.micropython.org/en/v1.15/library/uhashlib.html)

`MD5 - A legacy algorithm, not considered cryptographically secure. Only selected boards, targeting interoperability with legacy applications, will offer this.` 

I think this is another board supporting issue.

MATTYGILO, 2022-12-13:
Hi @y3k00000 have you yet gotten this to work, I tried this repo back in the day. It seemed to not work at all even if you compiled it and put it on a device?

y3k00000, 2022-12-14:
@MATTYGILO After the u2f was built you'll need to manually copy files from [Here](https://github.com/dmazzella/ucrypto/tree/master/modules) into the board's /lib directory.

If hashlib on your board don't provide the md5() you'll have to edit the py files and comment out all md5() relatives.

Then you can begin importing the module and see if it's working.
[image]

dmazzella, 2022-12-14:
solved in [eba2c08](https://github.com/dmazzella/ucrypto/commit/eba2c080352a76e912dc8a057c86c01bb1c46259)

dmazzella, 2022-12-14:
@y3k00000 if you can confirm that it works we can close the issue, tnx

y3k00000, 2022-12-14:
@dmazzella The importing is ok now. 👍
[image]

But when I tried running ecdsa test
[image]

It seems a further fix is required.

dmazzella, 2022-12-14:
fixed in [ffbd73](https://github.com/dmazzella/ucrypto/commit/ffbd73e0241b9c8760a1be335b168e0ae9c41671)

y3k00000, 2022-12-16:
Tested and it's ok now.

closed 2022-12-16 by y3k00000