dmazzella/ucrypto #11: overflow ecdsa

pedroelbanquero, 2022-12-29:
[image]

Hello I have this problem in esp32 , do you know how to solve ?

dmazzella, 2022-12-29:
Hi @pedroelbanquero, currently I haven't a esp32, what board and chip you use?

dmazzella, 2022-12-29:
can you try to isolate the problem in the [ecdsa.sign](https://github.com/dmazzella/ucrypto/blob/6111ee5fbecd1fd43dda4cd035866da4ea1d4988/modules/ufastecdsa/ecdsa.py#L23) function by running the steps one by one to identify whether the problem is in _crypto.ECC.ecdsa_sign or elsewhere?

pedroelbanquero, 2022-12-29:
Im using this chip ESP32 https://micropython.org/download/esp32/ 4MB flash version

Seems all working fine less the signature .

[image]

[image]

let me know how to pass correct params directly to _crypto . can u help me ?

pedroelbanquero, 2022-12-29:
seems crashing here in ecdsa lib

```
k = nonce or RFC6979(msg, d, curve.q, hashfunc=hashfunc).gen_nonce()

```

dmazzella, 2022-12-30:
i have tried your test but I can't replicate your error on stm32 board
```
s = _crypto.ECC.ecdsa_sign(b"affffffffffff", 23, 23, curve.P256._curve)
print(s)
```

```
micropython (master) ✗ python3 -Bu tools/pyboard.py -d /dev/cu.usbmodem3688376D31302 ports/stm32/boards/ARDUINO_PORTENTA_H7/modules/ucrypto/tests/ufastecdsa_1.py
<Signature r=6589947112124193829845813048643357708501113701986593856139712262332079569046 s=92175404354561421176187404272118520752411906693739112370103990833392490376502>
```

pedroelbanquero, 2022-12-30:
i have no idea how to solve, any idea i can test for you ?

pedroelbanquero, 2022-12-30:
i going to try other board with esp32

pedroelbanquero, 2022-12-30:
[image]

same with other esp32 board

pedroelbanquero, 2022-12-30:
[image]

this seems working fine

pedroelbanquero, 2022-12-30:
[image]

this method works

dmazzella, 2022-12-30:
Can you try to add some printf into [ecdsa_s](https://github.com/dmazzella/ucrypto/blob/6111ee5fbecd1fd43dda4cd035866da4ea1d4988/moducrypto.c#L1204)?

pedroelbanquero, 2022-12-30:
I think is ram issue, how many ram have your micro controller ?

I can add but no idea how to print there , can you give me the line you want to add ?

dmazzella, 2022-12-30:
try to reduce [2048](https://github.com/dmazzella/ucrypto/blob/6111ee5fbecd1fd43dda4cd035866da4ea1d4988/tomsfastmath/tfm_mpi.h#L128) to 512, if you need only ECDSA can be enough

pedroelbanquero, 2022-12-30:
where I change that ? RSA 2048 works fine .

dmazzella, 2022-12-30:
> I think is ram issue, how many ram have your micro controller ?
> 
> I can add but no idea how to print there , can you give me the line you want to add ?

```
STATIC void ecdsa_s(ecdsa_signature_t *sig, unsigned char *msg, size_t msg_len, fp_int d, fp_int k, const ecc_curve_t *curve)
{
    fp_int e, kinv;

    // R = k * G, r = R[x]
    ecc_point_t R;
    printf("ecdsa_s 0\n");
    ec_point_mul(&R, &curve->g, k, curve);
    printf("ecdsa_s 1\n");
    fp_init(&sig->r);
    fp_copy((fp_int *)&R.x, (fp_int *)&sig->r);
    printf("ecdsa_s 2\n");
    fp_mod((fp_int *)&sig->r, (fp_int *)&curve->q, (fp_int *)&sig->r);
    printf("ecdsa_s 3\n");

    // convert digest to integer (digest is computed as hex in ecdsa.py)
    fp_read_radix(&e, (const char *)msg, 16);
    printf("ecdsa_s 4\n");

    int orderBits = fp_count_bits((fp_int *)&curve->q);
    int digestBits = msg_len * 4;

    if (digestBits > orderBits)
    {
        fp_int n;
        fp_init(&n);
        fp_2expt(&n, digestBits - orderBits);
        fp_div(&e, &n, &e, NULL);
    }

    // s = (k^-1 * (e + d * r)) mod n
    fp_init(&kinv);
    fp_invmod(&k, (fp_int *)&curve->q, &kinv);
    fp_init(&sig->s);
    fp_mul(&d, (fp_int *)&sig->r, (fp_int *)&sig->s);
    fp_add((fp_int *)&sig->s, &e, (fp_int *)&sig->s);
    fp_mul((fp_int *)&sig->s, &kinv, (fp_int *)&sig->s);
    fp_mod((fp_int *)&sig->s, (fp_int *)&curve->q, (fp_int *)&sig->s);
}
```

pedroelbanquero, 2022-12-30:
step 0 

[image]

pedroelbanquero, 2022-12-30:
i was trying to enable SPI PSRAM over the flash without success for now ...

dmazzella, 2022-12-30:
Try add before first printf:

fp_init(&R.x);
fp_init(&R.y);

pedroelbanquero, 2022-12-30:
[image]

[image]

same

pedroelbanquero, 2022-12-31:
i make a solution in python just for the signature seems working fine, but overflows anyway when numbers are big

[image]

trying to solve redirecting memory to the flash .

dmazzella, 2023-01-01:
Can you print memory info with micropython.mem_info(1)?

pedroelbanquero, 2023-01-01:
```
micropython.mem_info(1)
stack: 720 out of 15360
GC: total: 108032, used: 74480, free: 33552
 No. of 1-blocks: 949, 2-blocks: 189, max blk sz: 210, max free sz: 2085
GC memory layout; from 3ffe53a0:
00000: MDhhhhBMMDDhhhh=hhh=================Th=======h=======h=hh=======
00400: ========BhhShShhhSFhhh===============h=======h==========BTMDSShS
00800: hhShShh===============hBSShMDFhSSh=hSh==hhSSh==Th=Bhhh=Sh=h=====
00c00: ==========hhhh=======BTBhhT=BBBDhh==B==h=h=Bh==h====BBTBBSh=====
01000: ==h==hh==T=TB=B=BBB=h==================hhSh=h========h=STh===MDS
01400: Shh=======h=hhBhhhhSh=h==Sh=ShSh=====h===============BBThhB=TSSh
01800: hhSB=h=Sh=======h====MDShhShShh=======h=h===============Bhhhhhh=
01c00: hSh=======h=======hBDhSh=======MDhhhh=Shhh=BTMhh=h==============
02000: =FShDShhSh=hhT=TSh=hhhhhhShSFBTBDTSh=Sh=======h==hSh=SShShSSh=h=
02400: hhhh===h==h===============hh====================================
02800: Sh=SSh===h=h=Shhh==hhSh=Shh====hh====hMDhhhSBBTTBBh==MhhShh=====
02c00: ==ShShShShh==========B=DhhhSh=hh==================Sh=BShhThBBShh
03000: =Shhhhh===============h====Thhhhhh=======Shh====================
03400: ===h===============================Sh=h====h====hShhhh=hhhB=BhBh
03800: hhh=======h===h=hh==hShShhhh=hhh==hhhhhhhhhh=hhh=====h==========
03c00: =====Sh========h==h===T=Mh=hhhhh==hh==h=hhh=BTTh=====Bhhh==h=h=h
04000: ==hh=h==hhT=h==B=Bhhh====hh====hh==hhhhhh===hhh=BBBhhhh=hhhhh==h
04400: h=h=BDTBhhh===hhhh==============================================
04800: =========================h=====h=======h========================
04c00: =h==h========h==========h===h==hhhh=========h==h=========h====h=
05000: h===h===h=======================================================
05400: ============================h=====================h==hh==MDShhSh
05800: Shhhhhh====h======h====h=====h====h=Bhhh========hhB=hhh=========
05c00: =========h========h========h======h========h===========h==h=====
06000: ======h===hh============h===========h====h=====hh=====hh=======T
06400: =h==================h=====h===h==Bh=h===========================
06800: ===h================================================h===========
06c00: =======h===============B=BBDSh========Shhh===h===h===hhh=Sh=h===
07000: ============BMDhShhSh=ShhTMh=h===============hBhhhhhhBDShhTBh=h=
07400: ==BDBTB=BBhBBhTh===h====MDSDhh===hhSShh=====hhh=============hhhh
07800: h===h=======h=h=hhhh=====h=h===h=====h=hhThhh=====h=====h=====T=
07c00: =ShBTMhhDSSDShhShhhShhh=Shh==SShhShhhhh=====h===============h=h=
08000: ==============h=======hBShhhhhhhSShhShShShhhh===h===h=======h==h
08400: ===h====h====ShShSSShhh====h====h====h====MDBTT=MDBDBBShSSSShh=h
08800: h===============Sh=====h=h===h=hShSh==h=h=hh==h============hh===
08c00: h====h=======BBBhhhhhh==hShh=====h=====h=h=hhhh=h===h=h====hh===
09000: ===h=hh==hhh=h=============================h=============h======
09400: =================h==============================================
09800: ================================================================
09c00: ================================================================
0a000: ===================================h=======hhh===ShhShh===SDTBTB
0a400: =ThSh==BBB=BBTB=SSh=h==SSSSSh=======h====hhhhBhhhh===hh=hh===hhh
0a800: h====h====BBh===h==========================================h====
0ac00: DBTB=h========h=======h============h=======h=======hh===h===hhh=
0b000: ============================BBhh=BTSh=h=======hh=BDTh===h===h===
0b400: h===h================TMDSh=h=hhh================================
0b800: ================================================================
0bc00: =============================================h==================
0c000: =========hhh=======BBDhhTTBSSBh=h=h===============hhhShShhShShhh
0c400: h=h=======hh================h===DBTB=hBBBBBBBBBhBhBhTB=Th=h=====
0c800: ===Sh=h===h===h===BDTBThBh=hBDhhhSShBhh==BBBhBhBThh==h==Bh=h===h
0cc00: ===h=h==h===============hh==T=h=h==hhh=====h======h===h==h===h==
0d000: =====h=h=======================h==================h==h=h=hhhhSSh
0d400: ===B=MDhSB=h=hhShhh===h========h=======h=h=BTThhhTTB=ThB=hhLh=B=
0d800: h==h===========h===============h====h====h=======Sh=============
0dc00: ========BBBBBBBBBBBBBBBhhh=hSSh=h=h=hh===============h=Sh=SSh=h=
0e000: ==h=====h=h=hh==================================================
0e400: ====h==============h=========h=hhh====hBBT=B=Bh=h=SShhBBBBBBBMDS
0e800: hBShShSh=h====h=h============h=h=====================h=h=h=h=h=h
0ec00: =h=h========hTThh=h===h=Sh=Sh=Sh================================
0f000: ====h=h=Sh====h======h=h==h=h=Shh=hhSSh=Sh=SSShh==h=hSh=SShh====
0f400: =======h=======h=h=SShh===========TSSh===h===h=======h========h=
0f800: h=hh===hh==h=hhhhhh=Sh=h=============================h=====h====
0fc00: ===h=BDMDBBTShhBDhBTB=hThBBShhh==hhh=======h===============h====
10000: ====h==============Dh=======hBBhBBBBBBTBDhh====BTSh============h
10400: =h=hShSShhh==h===hhB=hh=============TBBh==h==h===========h===h==
10800: =hh=h======h==TTBTBSh=========================Dh==h=============
10c00: =================B=hh==h=hhTh===h===BDhTh===Bh===h==h==h==h==hhh
11000: h=Bh=======h==h========h==============h===BBBBBhh===============
11400: ========h=h===BDhTBhh=BBBh========h===BBh=======================
11800: ======h=====ShT=h=hhh=======h=h=hh=======SBBBBBBBBBh============
11c00: ========================h===============================hThhh===
12000: ==========================h=hhBh=hhhhB..hhh=.......h=...h==.....
       (32 lines all free)
1a400: ................................

```

dmazzella, 2023-01-02:
try to use gc.threshold to reduce memory fragmentation:

```
import gc
gc.threshold(512)
```

dmazzella, 2023-01-02:
do you have freeze ucrypto modules or you use it as source code?

try to add `include("./modules/ucrypto/modules/manifest.py")`

into `ports/esp32/boards/manifest.py`

Fixing the path where you cloned the ucrypto sources

pedroelbanquero, 2023-01-02:
> try to use gc.threshold to reduce memory fragmentation:
> 
> ```
> import gc
> gc.threshold(512)
> ```

[image]

pedroelbanquero, 2023-01-02:
> do you have freeze ucrypto modules or you use it as source code?
> 
> try to add `include("./modules/ucrypto/modules/manifest.py")`
> 
> into `ports/esp32/boards/manifest.py`
> 
> Fixing the path where you cloned the ucrypto sources

same with that .

dmazzella, 2023-01-09:
try last commit if you can, it reduce memory usage

pedroelbanquero, 2023-01-09:
let me try !! thanks

pedroelbanquero, 2023-01-09:
[image]

crashing yet

dmazzella, 2023-01-09:
can you please add a `gc.collect()` and  `micropvthon.mem info(1 )`  before `r, s = ecdsa.sign(m, private _key)`?

pedroelbanquero, 2023-01-09:
```

>>> from ufastecdsa import curve, ecdsa, keys, util
>>> private_key, public_key = keys.gen_keypair(curve.P256)
>>> import micropython
>>> gc.collect()
>>> micropython.mem_info(1)
stack: 720 out of 15360
GC: total: 108032, used: 28928, free: 79104
 No. of 1-blocks: 183, 2-blocks: 49, max blk sz: 279, max free sz: 1764
GC memory layout; from 3ffe53a0:
00000: MDhhhh=MMDDhhhh=Bhh=================hh=======h=======h=h........
00400: ..h=..h==h===h==........MD.Sh.................................MD
00800: Sh.....h=======h==========...M....................hh.....DSh..Bh
00c00: =h===.DB.B=BBhBBh.h===h====.hh...Dhh===Sh..h=.h====h===.........
01000: ..................SS..h..................h==T==...........DS.h..
01400: .S..h=====................h...............................MD....
01800: .......h....h.B.....h..S.h.............h====......hh.hh====hh===
01c00: =hh====hh====Sh........S..h=======h==.................h.........
02000: ...h=hSh.h=h===h====h=======h==h=........................h======
02400: ================================================================
02800: ================================================================
02c00: ================================================================
03000: ===========......S...Sh==..........SD.B.B=....Sh..........Sh==S.
03400: .S....h..hh........h================BBB=hBBh===.B=h===h====h====
03800: B...h===...........................................h===B.D..h...
03c00: ......h=======h============h=======........h........h=..........
04000: ............................h=======.h=h================Bh==.B=B
04400: BB...h===h===h===.D..MDS.MDh.....................h==============
04800: ====..D....BSS.ShShh...h..h=h===DB.B=hBBB.hh===..hBBBBhBhBhB....
04c00: ........................Bh========.B=.h===h===.D.B..............
05000: ..........h=..D..hSS.B.h==h==..h==.h==h==BBBhBhB.Bh===h==h===h=.
05400: .h=h===h=.....h==.h=.h....h======h===h===.......................
05800: ............h==================h=======B=....h===B=...h========h
05c00: ====............................B=B=h===.B=....h====............
06000: .h====.............................h............................
06400: ...............h=======...h=..h..h...................h........h.
06800: ......................................................h=========
06c00: h=h====.............................h=.............h=.......h=h=
07000: ===========h=h=====================h=h=h=h=h=h=h=h========......
07400: ........................................................h======.
07800: ...........h==..................................................
07c00: ....................h=h========h=hh=hh===hh==h=hhhhh............
08000: .......................h=====h=======...........................
       (7 lines all free)
0a000: .....................hh=........................................
       (7 lines all free)
0c000: ......................h=========================================
0c400: ================================================================
0c800: ================================================================
0cc00: ================================================================
0d000: =============================================hh.....hh====.....h
0d400: h====h.hh.hh....................................................
       (3 lines all free)
0e400: ..................................................h=============
0e800: ================================================================
0ec00: ================================================================
0f000: ================================================================
0f400: ================================================================
0f800: =========............hhh........................................
       (26 lines all free)
16400: ............................................................h.h=
16800: .B.....h========================================================
16c00: ================================================================
17000: ================================================================
17400: =========================h.....................................h
17800: =hh=====........................................................
       (10 lines all free)
1a400: ................................

```

pedroelbanquero, 2023-01-26:
[image]

hello, i add more psram and works fine, pure ram issue . !!

closed 2023-01-26 by pedroelbanquero

dmazzella, 2023-01-26:
I'm glad everything works and thanks for the feedback