dmazzella/ucrypto #14: make on Pi Pico RP2040 board

cockerfly, 2023-08-30:
Hi, friend!) 
What is the right command to make firmware on Pi Pico 2040 board? When i used this ->  make -j8 -C mpy-cross && make -j8 -C ports/stm32/ BOARD="PYBD_SF6" USER_C_MODULES="$(pwd)/ports/stm32/boards/PYBD_SF6/cmodules"  firmware.elf is not supported. And when i convert this firmware with ./elf2uf2 firmware.elf firmware.uf2  I get an error "ERROR: HARD-FLOAT not supported"

Please help me to understand this

dmazzella, 2023-08-30:
Hi, try this:

clone ucrypto into `RPI_PICO/cmodules/ucrypto`:
```bash
micropython (master) ✗ git clone https://github.com/dmazzella/ucrypto.git $(pwd)/ports/rp2/boards/RPI_PICO/cmodules/ucrypto
```

edit `ports/rp2/boards/manifest.py` at the and of file add:

```python
include("$(BOARD_DIR)/cmodules/ucrypto/modules")
```

build `RPI_PICO`:

```bash
micropython (master) ✗ make -j8 -C mpy-cross && make -j8 -C ports/rp2 BOARD=RPI_PICO USER_C_MODULES="$(pwd)/ports/rp2/boards/RPI_PICO/cmodules/ucrypto/micropython.cmake"
```

cockerfly, 2023-08-30:
> 

I did everything you said, in the end an error still appears

CMake Error: The source directory "/home/ldb/pico/pico-examples/build/micropython/ports/rp2/build-RPI_PICO" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:54: recipe for target 'all' failed
make: *** [all] Error 1
[image]

dmazzella, 2023-08-30:
Have you cloned a fresh micropython repo? have you initialized submodules?

`git submodule update --init --depth 1`

cockerfly, 2023-08-30:
> Have you cloned a fresh micropython repo? have you initialized submodules?
> 
> `git submodule update --init --depth 1`

Yes. I did everything according to the instructions on the main page of the repo. Specifically this command --> "git submodule update --init"

dmazzella, 2023-08-30:
Have you tried a clean build without ucrypto?

Try first a clean build and if it succeded include the ucrypto module

cockerfly, 2023-08-30:
> Have you tried a clean build without ucrypto?
> 
> Try first a clean build and if it succeded include the ucrypto module

I'll try like this, thanks for help))

cockerfly, 2023-08-31:
> > Have you tried a clean build without ucrypto?
> > Try first a clean build and if it succeded include the ucrypto module
> 
> I'll try like this, thanks for help))

Clean build also print error(
[image]

cockerfly, 2023-09-01:
> Have you tried a clean build without ucrypto?
> 
> Try first a clean build and if it succeded include the ucrypto module

The main cause of the error was ubuntu of the old version. When I installed 22.04, everything worked immediately. Thanks again for the tips and good luck!
The topic can be closed)

dmazzella, 2023-09-02:
glad you solved it, good luck to you

closed 2023-09-02 by dmazzella