dmazzella/ucrypto #16: Errors compiling for ESP32_GENERIC_C3

jonahbron, 2024-03-08:
I'm trying to build a Micropython firmware with this library included.  I have an environment set up for building Micropython and can get a firmware.bin file out by running

```bash
make -C ports/esp32 BOARD=ESP32_GENERIC_C3
```

However when I clone down [ucrypto](../../) and include it in the `USER_C_MODULES` argument as instructed, I get quite a lot of errors and the firmware fails to build.

```bash
make -j8 -C ports/esp32 V=1 BOARD=ESP32_GENERIC_C3 USER_C_MODULES="$(realpath ../ucrypto/micropython.cmake)"
```

Here is the last chunk of output, but there are quite a few errors, all seemingly coming from `ucrypto/moducrypto.c`

```
/home/jonah/micropython/ucrypto/moducrypto.c:2225:7: error: expected ';' before 'const'
 2225 | STATIC MP_DEFINE_CONST_DICT(mp_module_ucrypto_globals, mp_module_ucrypto_globals_table);
      |       ^
      |       ;
/home/jonah/micropython/ucrypto/moducrypto.c:302:12: warning: 'fp_pow3' defined but not used [-Wunused-function]
  302 | static int fp_pow3(fp_int *X, fp_int *E, fp_int *M, fp_int *Y)
      |            ^~~~~~~
cc1: some warnings being treated as errors
[96/97] Generating binary image from built executable
esptool.py v4.7.0
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
Generated /home/jonah/micropython/micropython/ports/esp32/build-ESP32_GENERIC_C3/bootloader/bootloader.bin
[97/97] cd /home/jonah/micropython/micropython/ports/esp32...rts/esp32/build-ESP32_GENERIC_C3/bootloader/bootloader.bin
Bootloader binary size 0x4330 bytes. 0x3cd0 bytes (48%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonah/micropython/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stderr_output_39396 and /home/jonah/micropython/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stdout_output_39396
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
make: Leaving directory '/home/jonah/micropython/micropython/ports/esp32'
```

Here's the full output:\
https://paste.sr.ht/~jonahbron/c48f525093b76f9481d9340163c03c66fb289442

Is this a known issue, does ESP32C3 not work?  Or am I doing something wrong with my compilation?

dmazzella, 2024-03-08:
From the MicroPython tip they just removed the define of STATIC, please see: 

[all: Remove the "STATIC" macro and just use "static" instead.](https://github.com/micropython/micropython/commit/decf8e6a8bb940d5829ca3296790631fcece7b21)

I haven't had time to update the libraries yet, I will do it soon

jonahbron, 2024-03-10:
Thank you so much for the quick response! There are still a lot of warnings in the output, but I was able to build the firmware and run _ucrypto functions in the REPL.

closed 2024-03-10 by jonahbron

jonahbron, 2024-03-10:
Also want to report that I did successfully use the `manifest.py` to freeze the Python portion of this library into the firmware and _was_ able to execute the EC code in the README example!  This is really great, thanks so much

IcedRooibos, 2024-06-16:
I get these errors: building for micropython v1.23.0 and potentially related to changes between new MPY versions? I don't recall having this problem earlier.

Seems like the pointer's types are different - long unisnged int vs unsigned int, if you have any suggestions on how to fix this I will try and find some time to fix and issue a PR (alternatively if you prefer to fix this yourself then of course that is fine too!)

```
In file included from /home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:34:
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:421:55: warning: initialization of 'void * (*)(size_t,  void * const*, mp_map_t *)' {aka 'void * (*)(unsigned int,  void * const*, struct _mp_map_t *)'} from incompatible pointer type 'void * (*)(mp_uint_t,  void * const*, mp_map_t *)' {aka 'void * (*)(long unsigned int,  void * const*, struct _mp_map_t *)'} [-Wincompatible-pointer-types]
  421 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_exptmod_obj, 3, mod_exptmod);
      |                                                       ^~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:421:55: note: (near initialization for 'mod_exptmod_obj.fun.kw')
  421 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_exptmod_obj, 3, mod_exptmod);
      |                                                       ^~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:536:62: warning: initialization of 'void * (*)(size_t,  void * const*, mp_map_t *)' {aka 'void * (*)(unsigned int,  void * const*, struct _mp_map_t *)'} from incompatible pointer type 'void * (*)(mp_uint_t,  void * const*, mp_map_t *)' {aka 'void * (*)(long unsigned int,  void * const*, struct _mp_map_t *)'} [-Wincompatible-pointer-types]
  536 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_generate_prime_obj, 1, mod_generate_prime);
      |                                                              ^~~~~~~~~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:536:62: note: (near initialization for 'mod_generate_prime_obj.fun.kw')
  536 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_generate_prime_obj, 1, mod_generate_prime);
      |                                                              ^~~~~~~~~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:567:56: warning: initialization of 'void * (*)(size_t,  void * const*, mp_map_t *)' {aka 'void * (*)(unsigned int,  void * const*, struct _mp_map_t *)'} from incompatible pointer type 'void * (*)(mp_uint_t,  void * const*, mp_map_t *)' {aka 'void * (*)(long unsigned int,  void * const*, struct _mp_map_t *)'} [-Wincompatible-pointer-types]
  567 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_is_prime_obj, 1, mod_is_prime);
      |                                                        ^~~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:567:56: note: (near initialization for 'mod_is_prime_obj.fun.kw')
  567 | static MP_DEFINE_CONST_FUN_OBJ_KW(mod_is_prime_obj, 1, mod_is_prime);
      |                                                        ^~~~~~~~~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
      |                                                                                                        ^~~~~~~~
[... cut ...]
/home/username/q-micropython/micropython/ports/esp32/cmodules/ucrypto/moducrypto.c:2186:49: note: (near initialization for 'point_obj.fun.kw')
 2186 | static MP_DEFINE_CONST_FUN_OBJ_KW(point_obj, 3, point);
      |                                                 ^~~~~
/home/username/q-micropython/micropython/py/obj.h:391:104: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_KW'
  391 |     {{&mp_type_fun_builtin_var}, MP_OBJ_FUN_MAKE_SIG(n_args_min, MP_OBJ_FUN_ARGS_MAX, true), .fun.kw = fun_name}
  ```

dmazzella, 2024-06-17:
thanks for the report, solved in [7e0ccac725afc773bd35e31332701ffa8ca1cb13](https://github.com/dmazzella/ucrypto/commit/7e0ccac725afc773bd35e31332701ffa8ca1cb13)

IcedRooibos, 2024-06-17:
Sweet, thank you!