dmazzella/ucrypto #17: Install steps on Beetle ESP32

KosBar49, 2024-04-27:
Hi!, 

Dear Author this is really good project!, I am interested how I can use it on my boad could you please provide the steps how to install it?

dmazzella, 2024-04-29:
HI,
have you tried this build steps? [esp32_generic](https://github.com/dmazzella/ucrypto#esp32_generic)

KosBar49, 2024-04-30:
Hi, yes I was not able to build it...

I have tried also this tutorial: https://lemariva.com/blog/2022/01/micropython-upgraded-support-cameras-m5camera-esp32-cam-etc

run: 
`make BOARD=ESP32_GENERIC USER_C_MODULES=$(pwd)/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake`
and end up with following error:

` will be added
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/mdns (1.1.0)
[2/2] idf (5.0.4)
-- Project sdkconfig file /home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig
Loading defaults file /home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined...
/home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:139 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
-- App "micropython" version: v1.23.0-preview-344-gb1ac266bb
-- Adding linker script /home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/bartosz/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including User C Module(s) from /home/bartosz/micropython/ports/esp32/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake
CMake Error at /home/bartosz/micropython/py/usermod.cmake:42 (include):
  include could not find requested file:

    /home/bartosz/micropython/ports/esp32/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake
Call Stack (most recent call first):
  esp32_common.cmake:22 (include)
  main_esp32/CMakeLists.txt:11 (include)

Found User C Module(s):
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main_esp32 mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/bartosz/esp/esp-idf/components/app_trace /home/bartosz/esp/esp-idf/components/app_update /home/bartosz/esp/esp-idf/components/bootloader /home/bartosz/esp/esp-idf/components/bootloader_support /home/bartosz/esp/esp-idf/components/bt /home/bartosz/esp/esp-idf/components/cmock /home/bartosz/esp/esp-idf/components/console /home/bartosz/esp/esp-idf/components/cxx /home/bartosz/esp/esp-idf/components/driver /home/bartosz/esp/esp-idf/components/efuse /home/bartosz/esp/esp-idf/components/esp-tls /home/bartosz/esp/esp-idf/components/esp_adc /home/bartosz/esp/esp-idf/components/esp_app_format /home/bartosz/esp/esp-idf/components/esp_common /home/bartosz/esp/esp-idf/components/esp_eth /home/bartosz/esp/esp-idf/components/esp_event /home/bartosz/esp/esp-idf/components/esp_gdbstub /home/bartosz/esp/esp-idf/components/esp_hid /home/bartosz/esp/esp-idf/components/esp_http_client /home/bartosz/esp/esp-idf/components/esp_http_server /home/bartosz/esp/esp-idf/components/esp_https_ota /home/bartosz/esp/esp-idf/components/esp_https_server /home/bartosz/esp/esp-idf/components/esp_hw_support /home/bartosz/esp/esp-idf/components/esp_lcd /home/bartosz/esp/esp-idf/components/esp_local_ctrl /home/bartosz/esp/esp-idf/components/esp_netif /home/bartosz/esp/esp-idf/components/esp_partition /home/bartosz/esp/esp-idf/components/esp_phy /home/bartosz/esp/esp-idf/components/esp_pm /home/bartosz/esp/esp-idf/components/esp_psram /home/bartosz/esp/esp-idf/components/esp_ringbuf /home/bartosz/esp/esp-idf/components/esp_rom /home/bartosz/esp/esp-idf/components/esp_system /home/bartosz/esp/esp-idf/components/esp_timer /home/bartosz/esp/esp-idf/components/esp_wifi /home/bartosz/esp/esp-idf/components/espcoredump /home/bartosz/micropython/ports/esp32/managed_components/espressif__mdns /home/bartosz/esp/esp-idf/components/esptool_py /home/bartosz/esp/esp-idf/components/fatfs /home/bartosz/esp/esp-idf/components/freertos /home/bartosz/esp/esp-idf/components/hal /home/bartosz/esp/esp-idf/components/heap /home/bartosz/esp/esp-idf/components/http_parser /home/bartosz/esp/esp-idf/components/idf_test /home/bartosz/esp/esp-idf/components/ieee802154 /home/bartosz/esp/esp-idf/components/json /home/bartosz/esp/esp-idf/components/log /home/bartosz/esp/esp-idf/components/lwip /home/bartosz/micropython/ports/esp32/main_esp32 /home/bartosz/esp/esp-idf/components/mbedtls /home/bartosz/esp/esp-idf/components/mqtt /home/bartosz/esp/esp-idf/components/newlib /home/bartosz/esp/esp-idf/components/n [... cut ...]
-- Configuring incomplete, errors occurred!
See also "/home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_274429 and /home/bartosz/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_274429
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1`

KosBar49, 2024-04-30:
@dmazzella could you tell me on which version of the esp-idf you build?

dmazzella, 2024-05-01:
@KosBar49
 esp-idf (8fbf4ba605) ✔ git describe --tags
v5.0.4

for build I have used this steps:

➜ projects cd micropython 
➜ micropython (master) ✗ git submodule update --init --depth 1
➜ micropython (master) ✗  git clone https://github.com/dmazzella/ucrypto.git ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto
➜ micropython (master) ✗ make -j8 -C mpy-cross && make -j8 -C ports/esp32/ BOARD="ESP32_GENERIC" USER_C_MODULES="$(pwd)/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake"

<details><summary><b>output of the build</b></summary>
<p>

```bash
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
idf.py -D MICROPY_BOARD=ESP32_GENERIC -D MICROPY_BOARD_DIR="/Users/damianomazzella/projects/micropython/ports/esp32/boards/ESP32_GENERIC"  -DUSER_C_MODULES=/Users/damianomazzella/projects/micropython/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake -B build-ESP32_GENERIC build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Executing action: all (aliases: build)
Running cmake in directory /Users/damianomazzella/projects/micropython/ports/esp32/build-ESP32_GENERIC
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=ESP32_GENERIC -DMICROPY_BOARD_DIR=/Users/damianomazzella/projects/micropython/ports/esp32/boards/ESP32_GENERIC -DUSER_C_MODULES=/Users/damianomazzella/projects/micropython/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake -DCCACHE_ENABLE=0 /Users/damianomazzella/projects/micropython/ports/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /opt/homebrew/bin/git (found version "2.44.0")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/damianomazzella/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/damianomazzella/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/damianomazzella/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/mdns (1.1.0)
[2/2] idf (5.0.4)
-- Project sdkconfig file /Users/damianomazzella/projects/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig
Loading defaults file /Users/damianomazzella/projects/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined...
/Users/damianomazzella/projects/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:139 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/damianomazzella/.espressif/python_env/idf5.0_py3.12_env/bin/python (found version "3.12.2") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "micropython" version: v1.23.0-preview-346-g64f28dc1e-
-- Adding linker script /Users/damianomazzella/projects/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/damianomazzella/projects/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including User C Module(s) from /Users/damianomazzella/projects/micropython/ports/esp32/boards/ESP32_GENERIC/cmodules/ucrypto/micropython.cmake
Found User C Module(s): usermod_ucrypto
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main_esp32 mbedtls mqtt newlib nvs_flash openthread partition_table per [... cut ...]
/Users/damianomazzella/.espressif/python_env/idf5.0_py3.12_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    23072  (    5600 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1708304  (  323312 remaining)
total                  1773840
```
</p>
</details>

dmazzella, 2024-05-04:
@KosBar49 
have you been able to build correctly?

dmazzella, 2024-05-08:
@KosBar49 reopen the issue if you need

closed 2024-05-08 by dmazzella