dmazzella/ucrypto #19: Adding as usermodule for ESP8266

KosBar49, 2024-07-24:
Hi @dmazzella, 

I tried to make your library working with ESP8266, but got error while make command: 

`bash-4.4$ make -i USER_C_MODULES=../../examples/usercmodule
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../examples/usercmodule/cexample
Including User C Module from ../../examples/usercmodule/ucrypto
Including User C Module from ../../examples/usercmodule/subpackage
Including User C Module from ../../examples/usercmodule/cppexample
CC ../../examples/usercmodule/ucrypto/moducrypto.c
../../examples/usercmodule/ucrypto/moducrypto.c: In function 'mpz_as_fp_int':
../../examples/usercmodule/ucrypto/moducrypto.c:209:5: error: passing argument 4 of 'mpz_as_bytes' makes integer from pointer without a cast [-Werror]
     mpz_as_bytes(i, true, bsize, ib);
     ^
In file included from ../../py/objint.h:29:0,
                 from ../../examples/usercmodule/ucrypto/moducrypto.c:36:
../../py/mpz.h:149:6: note: expected 'size_t' but argument is of type 'byte *'
 bool mpz_as_bytes(const mpz_t *z, bool big_endian, bool as_signed, size_t len, byte *buf);
      ^
../../examples/usercmodule/ucrypto/moducrypto.c:209:5: error: too few arguments to function 'mpz_as_bytes'
     mpz_as_bytes(i, true, bsize, ib);
     ^
In file included from ../../py/objint.h:29:0,
                 from ../../examples/usercmodule/ucrypto/moducrypto.c:36:
../../py/mpz.h:149:6: note: declared here
 bool mpz_as_bytes(const mpz_t *z, bool big_endian, bool as_signed, size_t len, byte *buf);
      ^
cc1: all warnings being treated as errors
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: [../../py/mkrules.mk:90: build-ESP8266_GENERIC/ucrypto/moducrypto.o] Error 1 (ignored)
cp: cannot stat 'build-ESP8266_GENERIC/ucrypto/moducrypto.d': No such file or directory
/bin/sh: build-ESP8266_GENERIC/ucrypto/moducrypto.d: No such file or directory
LINK build-ESP8266_GENERIC/firmware.elf
xtensa-lx106-elf-ld: cannot find -lstdc++
make: [Makefile:222: build-ESP8266_GENERIC/firmware.elf] Error 1 (ignored)
xtensa-lx106-elf-size: 'build-ESP8266_GENERIC/firmware.elf': No such file
make: [Makefile:223: build-ESP8266_GENERIC/firmware.elf] Error 1 (ignored)
Create build-ESP8266_GENERIC/firmware.bin
esptool.py v1.2
readelf: Error: 'build-ESP8266_GENERIC/firmware.elf': No such file
xtensa-lx106-elf-objcopy: 'build-ESP8266_GENERIC/firmware.elf': No such file
Traceback (most recent call last):
  File "/tools/xtensa-lx106-elf/bin/esptool.py", line 1270, in <module>
    main()
  File "/tools/xtensa-lx106-elf/bin/esptool.py", line 1183, in main
    operation_func(args)
  File "/tools/xtensa-lx106-elf/bin/esptool.py", line 924, in elf2image
    data = e.load_section(section)
  File "/tools/xtensa-lx106-elf/bin/esptool.py", line 547, in load_section
    os.remove(tmpsection)
OSError: [Errno 2] No such file or directory: '/tmp/tmpK7XVJ1.section'
make: [Makefile:216: build-ESP8266_GENERIC/firmware.bin] Error 1 (ignored)
Traceback (most recent call last):
  File "makeimg.py", line 17, in <module>
    with open(sys.argv[1], "rb") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'build-ESP8266_GENERIC/firmware.elf-0x00000.bin'
make: [Makefile:217: build-ESP8266_GENERIC/firmware.bin] Error 1 (ignored)`

is there any chance that you may help me with this?

dmazzella, 2024-07-24:
Hi @KosBar49, fixed in [last commit](https://github.com/dmazzella/ucrypto/commit/eb80a53fd385ecf1db64e89906159cb33b93572c)

KosBar49, 2024-07-24:
Hi @dmazzella, many thanks now the issue is gone and see new one with: 

> CC hspi.c
CC modespnow.c
AS gchelper.s
CC ../../shared/libc/__errno.c
CC ../../shared/libc/string0.c
CC ../../shared/netutils/netutils.c
CC ../../shared/readline/readline.c
CC ../../shared/runtime/interrupt_char.c
CC ../../shared/runtime/pyexec.c
CC ../../shared/runtime/stdout_helpers.c
CC ../../shared/runtime/sys_stdio_mphal.c
CC ../../shared/timeutils/timeutils.c
CC ../../lib/libm/acoshf.c
CC ../../lib/libm/asinfacosf.c
CC ../../lib/libm/asinhf.c
CC ../../lib/libm/atan2f.c
CC ../../lib/libm/atanf.c
CC ../../lib/libm/atanhf.c
CC ../../lib/libm/ef_rem_pio2.c
CC ../../lib/libm/erf_lgamma.c
CC ../../lib/libm/fmodf.c
CC ../../lib/libm/kf_cos.c
CC ../../lib/libm/kf_rem_pio2.c
CC ../../lib/libm/kf_sin.c
CC ../../lib/libm/kf_tan.c
CC ../../lib/libm/log1pf.c
CC ../../lib/libm/math.c
CC ../../lib/libm/nearbyintf.c
CC ../../lib/libm/roundf.c
CC ../../lib/libm/sf_cos.c
CC ../../lib/libm/sf_erf.c
CC ../../lib/libm/sf_frexp.c
CC ../../lib/libm/sf_ldexp.c
CC ../../lib/libm/sf_modf.c
CC ../../lib/libm/sf_sin.c
CC ../../lib/libm/sf_tan.c
CC ../../lib/libm/wf_lgamma.c
CC ../../lib/libm/wf_tgamma.c
CC ../../lib/libm/ef_sqrt.c
CC ../../drivers/bus/softspi.c
CC ../../drivers/dht/dht.c
LINK build-ESP8266_GENERIC/firmware.elf
xtensa-lx106-elf-ld: cannot find -lstdc++
make: *** [Makefile:222: build-ESP8266_GENERIC/firmware.elf] Error 1

is this anything with the toolchain? to build I'm trying to leverage the docker image: larsks/esp-open-sdk that they described in: https://github.com/micropython/micropython/blob/master/ports/esp8266/README.md

dmazzella, 2024-07-24:
Hi @KosBar49, i haven't tried build for ESP8266, I don't use this chip and I haven't never build for it, sorry on this I can't help you

dmazzella, 2024-07-24:
probably the problem is in this example: cppexample

`Including User C Module from ../../examples/usercmodule/cppexample`

KosBar49, 2024-07-24:
I reached to the point where the above error is solved but now I am facing memory overflow issues, so probably I need to build only with selected modules: 

> CC build-ESP8266_GENERIC/frozen_content.c
LINK build-ESP8266_GENERIC/firmware.elf
xtensa-lx106-elf-ld: build-ESP8266_GENERIC/firmware.elf section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 25316 bytes
make: *** [Makefile:223: build-ESP8266_GENERIC/firmware.elf] Error 1

dmazzella, 2024-07-25:
the board you are using how much flash does it have?

KosBar49, 2024-07-25:
the output of the esptool.py --port /dev/ttyUSB0  flash_id is pointing to 4MB:

> Detected flash size: 4MB

dmazzella, 2024-07-25:
Try with `BOARD_VARIANT=FLASH_1M`

KosBar49, 2024-07-25:
try with this option the build is a little bit smaller, but still got an error: 

> CC ../../drivers/dht/dht.c
CC build-ESP8266_GENERIC-FLASH_1M/frozen_content.c
LINK build-ESP8266_GENERIC-FLASH_1M/firmware.elf
xtensa-lx106-elf-ld: build-ESP8266_GENERIC-FLASH_1M/firmware.elf section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 19048 bytes
make: *** [Makefile:222: build-ESP8266_GENERIC-FLASH_1M/firmware.elf] Error 1

closed 2024-08-12 by dmazzella