dmazzella/ucrypto #20: [ESP32]: Building with TFM optimizations on XTENSA results in "dangerous relocation" linker errors.

RSC-Games, 2024-10-30:
I enabled optimizations for ECDSA and RSA and compiled the project. It builds fine without the optimizations but with them I get these linker errors (for every single function?)

RSC-Games, 2024-10-30:
NOTE: Using the compiler flag `-mtext-section-literals` did not help matters and made them far worse.
[image]

dmazzella, 2024-10-30:
XTENSA is not an arch supported for asm optimizations

dmazzella, 2024-10-30:
If you need optimizations, please switch to [ucryptography](https://github.com/dmazzella/ucryptography.git)

closed 2024-10-30 by dmazzella

RSC-Games, 2024-10-31:
Okay I probably should have checked that. I could write my own but that would take quite a while