dmazzella/ucrypto #22: Issue with RSA test in RP2 pico W

Machally, 2025-01-06:
First of all, congratulations on your work! I started a "free time" challenge to create a hardwallet based on the RP2 Pico W to eventually automate DeFi operations.

The ECDSA functions worked very well, and I successfully added full support for Ethereum-style cryptography, including SECP256K1, deterministic k-values using RFC6979, and recovery ID. With this, I was able to sign transactions and send them via USB or air-gap them using a QR code displayed on an SSD1306 screen.

Now, I want to delve deeper into enhancing security by adding RSA support, but I’m struggling with using ufastrsa. I compiled the firmware with the following specifications:

c
Copiar código
#define TFM_NO_ASM
//#define TFM_ECC192
//#define TFM_ECC224
#define TFM_ECC256
//#define TFM_ECC384
//#define TFM_ECC512
#define TFM_RSA512
#define TFM_RSA1024
#define TFM_RSA2048
However, when trying to run the RSA example from the README, the program gets stuck, and I need to forcefully restart the Pico. I also tried copying ufastrsa/ and functools.py to the lib/ directory, but it still doesn’t work.

I would appreciate any insights or suggestions. Thank you!

dmazzella, 2025-01-07:
have you tried without optimizations?

```c
#define TFM_NO_ASM

// #define TFM_ECC192
// #define TFM_ECC224
// #define TFM_ECC256
// #define TFM_ECC384
// #define TFM_ECC512
// #define TFM_RSA512
// #define TFM_RSA1024
// #define TFM_RSA2048
```

Machally, 2025-01-07:
Yes, I tried. The issue seems to be related to the genrsa method from ufastrsa.genprime (used to derive 'n', 'e', and 'd'), because I tested passing some valid RSA parameters directly, and it worked:

`from ufastrsa.rsa import RSA

def main():

    bits = 1024
    print("RSA bits", bits)
    rsaInit_n = 0xb4d92132b03210f62e52129ae31ef25e03c2dd734a7235efd36bad80c28885f3a9ee1ab626c30072bb3fd9906bf89a259ffd9d5fd75f87a30d75178b9579b257b5dca13ca7546866ad9f2db0072d59335fb128b7295412dd5c43df2c4f2d2f9c1d59d2bb444e6dac1d9cef27190a97aae7030c5c004c5aea3cf99afe89b86d6d
    rsaInit_e = 0x10001
    rsaInit_d = 0x........................8f24f42073b328847724fbbd293eee9cdec29ac4dfe953a4256d7e6b9abee426db3b4ddc367a9fcf68ff168a7000d3a7fa8b9d9064ef4f271865045925660fab620fad0aeb58f946e33bdff6968f4c29ac62bd08cf53cb8be2116f2c339465a64fd02517f2bafca72c9f3ca5bbf.....................
    r = RSA(bits=bits, n=rsaInit_n, e=rsaInit_e, d=rsaInit_d)
    if r:
        print("RSA OK")
        data = b"a message to sign and encrypt via RSA"
        print("random data len:", len(data), data)
        print(r.pkcs_verify(r.pkcs_sign(data)))
        assert r.pkcs_verify(r.pkcs_sign(data)) == data
        print("pkcs_verify OK")
        assert r.pkcs_decrypt(r.pkcs_encrypt(data)) == data
        print("pkcs_decrypt OK")

if __name__ == "__main__":
    main()`

    
--------LOGS--------
RSA bits 1024
RSA OK
random data len: 37 b'a message to sign and encrypt via RSA'
b'a message to sign and encrypt via RSA'
pkcs_verify OK
pkcs_decrypt OK
-------------------------

This worked both with and without optimizations.

dmazzella, 2025-01-09:
I am currently without boards with rp2 to check the problem, as soon as I have one in my hands I will come back to the problem.

dmazzella, 2025-01-27:
Hi @Machally, sorry for late return on this.

the problem seems to be in the `malloc` allocation function of rp2.

I attach a [.zip](https://github.com/user-attachments/files/18563873/tomsfastmath.zip) containing` tfm_mpi.c` and `tfm_mpi.h`, can you replace the ones in the `tomsfastmath` folder with these?

I await your feedback on this matter

dmazzella, 2025-01-28:
@Machally  solved in [094b9fa83a350494917f15a661563815987bf8f0](https://github.com/dmazzella/ucrypto/commit/094b9fa83a350494917f15a661563815987bf8f0)

closed 2025-01-28 by dmazzella